bldrs-ai/Share #1250: Implement Selection Sets with Persistent Visibility Control

OlegMoshkovich, 2024-08-05:
Often models contain elements that need to be persistently hidden to isolate specific portions of the model. 

Introducing the concept of selection sets would improve the layer of organization, allowing for a custom element structure and control over the visibility of elements. 
This feature would improve user workflow by enabling users to manage and share visibility settings of the model across sessions.

[Figma screens](https://www.figma.com/design/DqKVaQEukf8vVHnlwiRGhv/Share-Screens?node-id=1887-3901&t=5FMjqJx5Mzw5iUfR-1)

OlegMoshkovich, 2024-08-05:
https://technical.buildingsmart.org/standards/ifc/mvd/

OlegMoshkovich, 2024-08-05:
ChatGPT suggestion:

Yes, it is possible to hide elements in an Industry Foundation Classes (IFC) model using Model View Definitions (MVDs). MVDs define specific subsets of the IFC schema that are required for particular purposes or workflows, enabling the exchange of only the necessary information. Here's how you can use MVDs to hide elements in an IFC model:

Using MVD to Hide Elements in IFC
1. Define a Specific View
Create a specific view definition within the MVD that excludes the elements you want to hide. This involves specifying the required data and excluding the unnecessary elements.

2. Use Filtering Criteria
Apply filtering criteria to determine which elements should be hidden. This can be done based on certain attributes, types, or properties of the elements.

3. Custom Property Sets
Utilize custom property sets to mark elements for visibility control. For example, a property set could include a "Visible" boolean attribute that can be set to false for hidden elements.

4. Implement Exchange Requirements
Specify exchange requirements within the MVD to include only the relevant elements for a particular workflow or data exchange scenario. This helps in omitting elements that are not required.

Example Steps
Define the View Definition: Create a new MVD with a view definition that specifies which elements are included and excluded.
Apply Filtering: Use filtering criteria to identify elements that should be hidden. For example, you can filter by element type, such as excluding all IfcFurniture elements.
Custom Property Set: Add a property set to elements that includes a visibility attribute. For example:
xml
Copy code
<PropertySet>
    <Name>VisibilityControl</Name>
    <Properties>
        <Property>
            <Name>Visible</Name>
            <Type>Boolean</Type>
            <Value>false</Value>
        </Property>
    </Properties>
</PropertySet>
Configure Exchange Requirements: Set up exchange requirements in the MVD to ensure that only elements with Visible=true are included in the exchange file.
Tools and Implementation
IFC Documentation and Standards: Refer to the official IFC documentation and buildingSMART standards for detailed guidelines on creating and using MVDs.
MVD Authoring Tools: Utilize tools like the buildingSMART Data Dictionary or other MVD authoring tools to create and manage your MVDs.
IFC Export Settings: Configure your BIM software's IFC export settings to respect the visibility criteria defined in your MVD.
Summary
While IFC and MVDs do not have a direct "hide" functionality like some BIM software, you can achieve similar results through careful definition of view definitions, filtering criteria, custom property sets, and exchange requirements. By specifying what elements should be included or excluded in your MVD, you can control the visibility of elements in the resulting IFC file.

OlegMoshkovich, 2024-08-05:
**How to hide elements in the IFC file**

ChatGPT
Hiding elements directly in an IFC file typically involves modifying the properties or attributes of the elements so that they are not displayed by the software that reads the IFC file. Here are some methods to achieve this:

Methods to Hide Elements in IFC Files

Custom Property Sets:

Add a custom property set to the elements you want to hide. This property set could include a visibility attribute, such as a boolean Visible property.
Software reading the IFC file can be configured to check this property and hide elements accordingly.
Using Layer Assignments:

Assign elements to specific layers and control their visibility through layer settings. This is dependent on the software's ability to interpret and manage layer visibility.

Modifying Element Attributes:

Set specific attributes of elements to make them less prominent or hidden. For example, changing their geometric representation to an empty shape or modifying their IfcPresentationLayerWithStyle to have no visible representation.
IFC Property and Quantity Filtering:

Use specific IFC properties and quantities to filter elements during import/export processes. Elements with certain properties can be excluded based on predefined rules.
View Definition and Filtering:

Define view definitions (MVDs) that exclude certain elements. When exporting or viewing the IFC file, use these definitions to hide elements.

OlegMoshkovich, 2024-08-06:
Introducing the concept of selection sets would improve the layer of organization, allowing for a custom element structure and control over the visibility of elements. 
This feature would improve user workflow by enabling users to manage and share visibility settings of the model across sessions.

OlegMoshkovich, 2024-08-06:
**Schematic sequence** 
- Change the visibility of an element.
- Create a Bldrs property set in the element.
- Assign the visibility to false.
- Serve the message to the user along the lines of "To track the current state of the model, commit your changes when you are ready."
- Add an icon to the save dialog indicating there are unsaved changes.
- When the user opens the save dialog, it defaults to the save version.

If the user did not commit the changes:
- Before closing the model, ask the user if the changes should be committed.

pablo-mayrgundter, 2026-09-29:
Partly addressed by #1885, which encodes hide / isolate state in the permalink's `#d:` token (`hide=` / `show=` / `iso=`). It works for IFC (element ids), STEP (occurrence paths) and scene-graph formats such as ADF, OBJ and GLB (NavTree name paths). Opening a link restores what the sender had hidden or isolated. Details are in design/new/model-display-controls.md §6.3.

What's left for this issue is the part that doesn't go through a link: keeping hidden elements across sessions for the same user (e.g. stored per model), and the Figma flow linked above. I'm leaving it open for that.

---
_Generated by [Claude Code](https://claude.ai/code)_