bldrs-ai/Share #1375: Upload error - Stohr House IFC

russell-stohr, 2025-05-20:
Dragged and dropped model and clicked save to upload to testmodels repo while talking with Pablo.  Came across this error.

<img width="1995" alt="Image" src="https://github.com/user-attachments/assets/2ec9f8ef-b526-488c-8fa0-d1d5829494f7" /

pablo-mayrgundter, 2025-05-20:
I tried to repro and got a different error

[image]

pablo-mayrgundter, 2025-06-06:
Error is now shown to the user in #1384. 

This is a limit on the GH API for and so we can't handle this directly.  We discussed some alternatives for staging git files from the browser, but they're very involved.  Will have to see later.

The workaround is to use GitHub Desktop https://github.com/apps/desktop to upload the file to the repo.

Viewing the file works fine after it is hosted at GH.

closed 2025-06-06 by pablo-mayrgundter